nrf-rs/nrf-hal #14: Consider transitioning peripheral access crates to this organization

hannobraun, 2018-09-12:
There are two peripheral access crates that the crates in this repository depend on, but that are not currently living under this organization [nrf52](https://crates.io/crates/nrf52) and [nrf52840](https://crates.io/crates/nrf52840). Would the authors consider transitioning those crates to the nrf-rs organization?

If we do transition those repositories, nrf52 should be renamed to nrf52832.

cc @jamesmunns, @wez, @jscarrott

hannobraun, 2018-09-12:
If we decide to move these crates here, we should also consider merging them into one repository, as they can likely share build infrastructure (like that Bash script I made for nrf52) between them.

Prior art: https://github.com/adamgreig/stm32-rs

jamesmunns, 2018-09-13:
For now, I have moved the peripheral access crate to this organization. I will also rename it as @hannobraun suggested above.

Also :+1: from me for merging them as a separate task.

jamesmunns, 2018-09-13:
Blocked, as I cannot rename items as I don't have admin permissions for repos in this orga yet. CC @wez

jscarrott, 2018-09-13:
I can do that

hannobraun, 2018-09-13:
Awesome!

closed 2018-09-13 by hannobraun