nrf-rs/nrf-hal #15: Consider creating nrf52 GitHub team; giving this team access on crates.io

hannobraun, 2018-09-12:
We should consider creating an nrf52 team on GitHub under this organization, and give this team access to all related repositories and crates. This would keep access management concentrated in one place (GitHub), allow everyone to release new crate versions as necessary, and make it easier to tag the relevant people on any issues.

cc @jamesmunns @jscarrott @wez

wez, 2018-09-13:
OK, I've renamed the team to nrf52 and we can continue to assess and make changes as needed.

wez, 2018-09-13:
(still need to update the crates.io access)

jamesmunns, 2018-09-13:
I have granted crates.io access.

Because of this, I would like to require 2FA for all members of the nrf52 organization. I will create a separate issue for this.

closed 2018-09-13 by jamesmunns