nrf-rs/nrf-hal #16: Consider moving boards/adafruit_nrf52pro into a separate repository

hannobraun, 2018-09-12:
I'm a big fan of keeping things neat and tidy, and while there's a good argument to be made that all HAL-related crates should remain in this repository, I think we should maintain board support crates in a separate repository. I believe this would have the following advantages:
- Improve discoverability of the `adafruit_nrf52pro` crate. Currently it's pretty well-hidden in this repository.
- Ease maintenance of this repository. https://github.com/nrf-rs/nrf52/pull/6 is an example where the crate causes build errors. It's no big deal in this case, but I wouldn't want to force contributors to fix errors in a crate they might know nothing about, for a board they might not have available.

wez, 2018-09-13:
FWIW, I think it is valuable for one or two crates in this category to be in this repo so that it is clearer what impact we're having as we make changes in the core, and importantly, so that the CI can highlight build problems; Github doesn't have a good story for expressing dependent PRs, let alone coordinating with CI systems.

I don't think this repo should be the home for all such crates though, as that is definitely not scalable.

Is there a good solution for the dependency issue that I'm not aware of?

hannobraun, 2018-09-13:
@wez 
> FWIW, I think it is valuable for one or two crates in this category to be in this repo so that it is clearer what impact we're having as we make changes in the core, and importantly, so that the CI can highlight build problems

Makes sense.

> Is there a good solution for the dependency issue that I'm not aware of?

Not that I'm aware of.

Another argument for having board support in-tree is that it's beneficial to use a widely available board for examples (otherwise examples are either very limited or require custom circuits).

In any case, if we decide to keep the board crate, it makes sense to add another one for 52840, too.

hannobraun, 2018-09-13:
@wez Did you choose the Adafruit board for any particular reason, or was it just something you had available? Maybe it makes sense to support the official development kits from Nordic instead (haven't looked into that yet, just thinking out loud).

wez, 2018-09-13:
I have a couple of the adafruit boards to hand.  In the US the adafruit boards are seemingly more readily available than the nordic dev boards in my limited and possibly skewed experience.  They're also a bit cheaper and better suited (form factor and feature wise) for hobbyists to integrate into their builds.

Re: 52840, I just obtained one of these boards: https://wiki.makerdiary.com/nrf52840-mdk.  I picked that one up mostly because the nordic dev boards seemed to be unavailable from the handful of distributors I checked at the time, and this board (that I'm eyeing for a project) https://www.particle.io/mesh/buy/xenon isn't shipping yet.

jscarrott, 2018-09-13:
If it makes any bearing last weekend I pushed up some code for the nordic nRF52840-dk as that is the board I have been using. I was waiting for the dust to settle on the #5 before I was going to open a PR.

wez, 2018-09-14:
I added https://github.com/nrf-rs/nrf52840-mdk-rs in a repo of its own; I expect that this particular board will be pretty niche adoption wise, even though it is not a bad little board.

hannobraun, 2018-09-14:
Okay, I've changed my mind. I'm no longer opposed to have board support crates in this repo, as long as we don't overdo it.

@jamesmunns I noticed you +1'd my original comment. Are you okay with closing this issue?

jamesmunns, 2018-09-15:
I'm a fan of keeping things separate, but I do agree that if you need to push changes that affect all repos fairly often, it makes more sense to use the mini-mono-repo pattern.

Lets run with this for now, and if it gets painful, we can reconsider then.