nrf-rs/nrf-hal #9: Investigate implementing embedded-hal traits for TIMER API

hannobraun, 2018-09-12:
This issue was first opened in the [old nrf52-hal repository](https://github.com/jamesmunns/nrf52-hal). Original discussion: https://github.com/jamesmunns/nrf52-hal/issues/10

---

I didn't initially implement embedded-hal traits for the new TIMER peripheral API, as embedded-hal only has a `CountDown` trait, while TIMER counts up. Now that I looked into it, I don't think this is a problem, as there's no method to read back the timer value in the `CountDown` trait.

From the [embedded-hal documentation](https://docs.rs/embedded-hal/0.2.1/embedded_hal/timer/trait.CountDown.html):
> Note that the implementer doesn't necessarily have to be a downcounting timer; it could also be an upcounting timer as long as the above contract is upheld.

So it looks like the trait can be implemented.

hannobraun, 2019-02-21:
This has been implemented in https://github.com/nrf-rs/nrf52-hal/pull/55. Closing.

closed 2019-02-21 by hannobraun

Yatekii, 2019-02-21:
Awesome! Thank you!