nrf-rs/nrf-hal #18: Consider changing repository configuration regarding pull request merging

hannobraun, 2018-09-13:
When merging pull requests, the option "Create a merge commit" is currently disabled in this repository. I actually prefer that option, as it documents in the commit history which commits belong to larger pieces of work.

"Squash and merge" is currently enabled. I suggest disabling it, as that makes it easy to accidentally lose valuable history, especially since it's currently the default option. Whenever squashing commits is desirable, it's easy enough to do manually.

I'd also prefer to disable "Rebase and merge", due to my preference for merge commits. I don't feel strongly about this point.

cc @nrf-rs/team, especially @wez

jamesmunns, 2018-09-13:
I generally prefer merge commits as well. No opinion from me regarding squash and merge.

wez, 2018-09-13:
I agree that squash should go.  I personally prefer rebasing to keep history more linear (my day job is wrangling massive source control systems) but am fine with merging if that’s your collective preference.

hannobraun, 2018-09-13:
@wez 
> I agree that squash should go. I personally prefer rebasing to keep history more linear (my day job is wrangling massive source control systems) but am fine with merging if that’s your collective preference.

I don't feel too strongly, and I realize that linear history has its benefits. I think the main problem right now is that "Merge" is disabled, and "Squash" is at the top of the list. I'm perfectly fine with leaving "Rebase" enabled, and could even live with "Rebase" being the only option :)

wez, 2018-09-13:
I've turned off squash, which effectively makes rebase the "default".  I've left merge out for now, but please don't hesitate to enable it if you have a situation where you want to merge.

closed 2018-09-13 by wez