nrf-rs/nrf-hal #24: GPIO: Add support for tri-state mode

hannobraun, 2018-09-17:
Some use cases require switching a pin between output and floating mode on the fly. This is very inconvenient with the current API. The DWM1001 board support crate [has an example](https://github.com/braun-robotics/rust-dwm1001/blob/78eb5df98709d7d2425e3d943d4d80cd782086da/src/lib.rs#L638) that demonstrates this.

I believe the solution for this is to add a tri-state mode that implements the following trait:
``` rust
pub trait TriStatePin {
    fn set_high(&mut self);
    fn set_low(&mut self);
    fn set_floating(&mut self);
}
```
Such an abstraction has been [proposed before](https://github.com/rust-embedded/embedded-hal/pull/73#issuecomment-379591543).

If we decide to implement this, we should submit the trait I suggested above to embedded-hal.

Yatekii, 2019-02-06:
I see the use of this and would welcome an implementation.

Atm those functions do not alter the Pin type ofc, so forgive me the dumb question, but how can I, after, let's say a `::set_floating()` of a Pin<Output<MODE>> access a let's say `::is_low()`? Also the generic type of that pin would not represent the actual pin state anymore after such a proposed `TriStatePin` operation if I understand everything right.

So do you think that is of no concern or do I missunderstand you?

If I understand everything correctly we maybe should rather return a converted Pin from the `TriStatePin` methods and make them consume self?

hannobraun, 2019-02-07:
I think we'd need a separate tri-state mode that is distinct from input or output mode. So you'd call `into_tri_state_pin`, which gets you a `Pin<TriState>`, which would then make the methods above available.

The lack of compile-time state tracking within the tri-state mode is intended, and indeed, is the whole point of why I suggested this. See the example I linked for an example of the compile-time state tracking getting into the way.

The methods I propose only cover the use case from my example, which doesn't require `is_low`/`is_high`. A complete implementation would include those within the tri-state mode.