nrf-rs/nrf-hal #33: Current Use of EasyDMA can cause incorrect optimization

jamesmunns, 2018-11-11:
Hey, I was doing some work with short EasyDMA TWIM transactions, and noticed that certain operations failed in release (opt level 3), but not in debug (opt level 0). Furthermore, they worked in opt levels 0, 1, or 'z', but not in opt level 2 or 3.

I believe this is due to the compiler not "understanding" our use of giving buffers as `.ptr()`s, and assuming that certain data is never used.

I was able to fix this by placing a compiler fence (`compiler_fence(Ordering::AcqRel);` after the DMA transaction was completed. It is likely we will need to use this everywhere EasyDMA is used.

jamesmunns, 2018-11-11:
CC @nrf-rs/nrf52

jamesmunns, 2018-11-11:
Relevant IRC logs, including some ASM diffs of what the compiler fence changes: https://mozilla.logbot.info/rust-embedded/20181111#c15597073-c15597632

hannobraun, 2018-11-12:
Good catch, @jamesmunns! Jorge wrote about the need for compiler fences in DMA code in his original blog post on the topic. I included that in my DMA implementation for lpc82x-hal, but unfortunately I didn't remember any of it while working on nRF52.

I think just adding the compiler fences everywhere they're needed is an acceptable solution for right now. But aside from that, I believe we shouldn't update or add any further DMA code before cleaning up what we currently have.

I think we've already talked about this on various pull requests, but I believe there should be a mid-level API (between the low-level PAC and the high-level HAL) that encapsulates things like task registers, event registers, and (most pressingly) DMA registers. Then all HAL APIs would use the mid-level API for DMA and we would have a single place to fix this issue.

jamesmunns, 2018-11-12:
Hanno,

That sounds fine to me, I have added one more method to TWIM locally, which was necessary to support the i2c accelerometer. I haven't PRd that yet, but you can [see it here](https://gist.github.com/jamesmunns/fecc1a0adbe15ffa76cf8f2d268ba84d).

I'm happy to keep it local for now, until the easydma interface has been reworked.

hannobraun, 2018-11-12:
@jamesmunns 
> That sounds fine to me, I have added one more method to TWIM locally, which was necessary to support the i2c accelerometer. I haven't PRd that yet, but you can [see it [...]

I mean, if you want to merge that one (since it's already written), I won't veto that. I just think we need to stop piling up the duplicated DMA code at some point :)

jamesmunns, 2018-12-10:
Closed by #34

closed 2018-12-10 by jamesmunns