nrf-rs/nrf-hal #54: Add resources for the two nrf52 chips to the README/repo

Yatekii, 2019-02-06:
I think it would be useful to add resources such as reference manuals to the repo.

Question is if it should be links to the PDFs in the README or even included PDFs in the repo that are omitted in a cargo publish.
I would really welcome that as google et. al always show prelim. versions as first hits, potentially fooling users (e.g the 52840er chip had 255byte wide EDMA transfer buffers for 2/3 peripherals when in fact it really has 2^16 - 1 bytes wide transfer buffers for all peripherals (at least at release (v1.0))).

Adding just a link keeps the storage footprint of the repo low and doesn't clutter it so much, where as committing the actual file would prevent broken links in case nordic moving stuff (which they seem to do some time; didn't happen for the reference manual yet tho; afaik).

What do y'all think about this?

Including [0, 1] would up the repo size about 15 Megs.

[2] http://infocenter.nordicsemi.com/pdf/nRF52840_PS_v1.0.pdf
[1] http://infocenter.nordicsemi.com/pdf/nRF52832_PS_v1.1.pdf

jamesmunns, 2019-02-06:
The 52832 is on PS 1.4 last I checked, btw

Yatekii, 2019-02-06:
Oh! Well that confirms my point :P

jamesmunns, 2019-02-27:
@Yatekii would you be interested in adding a table in the README for now linking to where these are officially linked? I'd say if we start running into breakage, we can decide to switch over, but I'd prefer not to check in binary items for now.

Yatekii, 2019-02-27:
Ok, will do!

PR #66 (2019-02-28): Add reference manual links to the README
Title.

Closes #54
closed 2019-02-28 by jamesmunns