nrf-rs/nrf-hal #56: Support for nRF52810?

chocol4te, 2019-02-08:
Hello!

If I started work on an `nrf52810-hal` is it the sort of thing you would like to see merged, or developed as a separate project? I can't see any references to it in this repository, so I was just wondering what the plans were.

Thanks :)

Yatekii, 2019-02-08:
Hi!

Sure, if you did an implementation and the code is okay, we'll happily merge that!
It seems that just nobody did have the need of an implementation yet. I also didn't hear much of the nRF52810 yet, so not sure how widespread it is.

You are welcome to contribute anytime!

hannobraun, 2019-02-09:
What @Yatekii said.

> I was just wondering what the plans were.

This is a pretty loosely organized group, so there aren't really any plans beyond what the individual contributors want. If you want to support 53810 in this crate, go ahead and make it happen :-)

One caveat: I'm not sure how many of the current maintainer have nRF52810 hardware and experience (I know I have neither). I'm guessing this would require some ongoing maintenance on your part, after the initial contribution.

eupn, 2019-02-10:
As far as hardware goes, nRF52810 can be "emulated" on nRF52832 dev. board, since former is just a stripped version of the latter. For that reason there is no nRF52-DK in existence which is specifically for the nRF52810 chip. That being said, the only support that this chip is requiring is a correct linker script which takes into account smaller amount of memories available and, perhaps, to implement feature-gating of peripherals that aren't present on the chip.

chocol4te, 2019-02-10:
@eupn Good to know, thank you! :)

chocol4te, 2019-02-10:
I'm not sure which is better of the following two options:

```
#[cfg(feature = "52810")]
use crate::target::{spim0, SPIM0};

#[cfg(any(feature = "52832", feature = "52840"))]
use crate::target::{spim0, SPIM0, SPIM1, SPIM2};
```

or

```
use crate::target::{spim0, SPIM0};

#[cfg(any(feature = "52832", feature = "52840"))]
use crate::target::{SPIM1, SPIM2};
```

Any thoughts?

Yatekii, 2019-02-10:
I think both are valid and okay, but I personally would go for the second.

The first one tells you on first sight what each chip can do. The second one tells you the differences.

From a style perspective I would go for the second one. But I think the first one is better when it comes to understanding code.

chocol4te, 2019-02-10:
I've gone with the second option and opened a PR [here (#57)](#57).

jamesmunns, 2019-02-11:
Closed by #57

closed 2019-02-11 by jamesmunns