nrf-rs/nrf-hal #74: Consider using .chunks() method for sending blocks of data

jamesmunns, 2019-03-07:
For sending multiple blocks of data for things like Spim::transfer, this is a handy pattern I came up with. Not sure if it compiles to more code:

```rust
fn send(&mut self, buf: &[u8]) -> Result<(), ()> {
    for c in buf.chunks(EASY_DMA_SIZE) {
        self.scratch[..buf.len()]
            .copy_from_slice(c);
        self.uart.write(&self.scratch[..buf.len()]).map_err(|_| ())?;
    }
    Ok(())
}
```

simonsso, 2019-03-09:
Interesting, there is also chunks_mut which would be more semantically correct for the if EasyDMS is filling the buffer.

In the example it looks like there is always a copy, I assume it is an illustration.

jamesmunns, 2019-03-09:
@simonsso yeah, I wrote this in a project that uses the 0.7.0 release, not `master`, so I always copy (for now). This should go away in the 0.8.0 release with all the recent changes :).

Good call on `chunks_mut`, which would make sense for the "read to a bufffer" half of the work :)

jamesmunns, 2019-03-11:
CC #76