nrf-rs/nrf-hal #82: SAADC: Multiple channels

chocol4te, 2019-03-21:
The SAADC can have multiple channels, and starting the `sample` task will result in all channels being read and DMA'd into memory. This is a useful feature and should probably be exposed somehow, but the specifics need to be discussed. 

It would most likely interfere with the embedded-hal `OneShot` trait, so perhaps a `MultiShot` trait could be proposed?

simonsso, 2019-03-22:
Because this feature is hardware dependent, it does not have comply with embedded_hal traits, however its naming conversions should be taken into account not to confuse the user. I don't think MultiShot is a descriptive name - I immediately assumed it would be about multiple samples, which is not the case.

It sounds more like a MultiChanelOneShot trait... or something to give the correct associations.

chocol4te, 2019-08-07:
`MultiChannelOneShot` sounds good to me, `MultiShot` is already a protocol in the RC world (and now also a Minecraft thing). I've been busy recently, but I'll try and put aside some time to work on this.

huntc, 2021-10-01:
Just wondering if there was more thought on this. Is the lack of activity here also related to the utility of multi channel?