nrf-rs/nrf-hal #109: Investigate implementing embedded-hal traits for UARTE

jonathanpallant, 2019-08-09:
I looked at implementing embedded-hal Serial traits for UARTE, but it wasn't immediately obvious how we might do that. All of my implementations so far have been with a hardware FIFO and so writing byte by byte makes sense. Sending a string of 100 bytes as 100 single-byte DMA operations would clearly be sub-optimal.

Has anyone given this any thought? Does it need to just wait for the interrupt-driven UARTE driver?

jonathanpallant, 2019-10-28:
Any thoughts on this one? @jamesmunns? @hannobraun?

hannobraun, 2019-10-30:
- I think we should implement the byte-by-byte trait, regardless of how bad that implementation is going to be. It's the standard interface, and a bad implementation is better than not supporting the respective drivers at all.
- `embedded-hal` also has higher-level blocking traits with an opt-in default implementation. For most other targets, that default implementation is pretty optimal, but in our case, we should implement that trait directly.
- I assume that at some point (once we have stable async/await, `async fn` in traits, and whatever else), it will make sense to add high-level async traits to `embedded-hal`. Once that happens, we should implement those too, and hope people start using is in favor of the stinking byte-by-byte implementation :-)

lulf, 2021-08-24:
I think this issue can be closed now that #343 was merged?

closed 2021-08-24 by jonas-schievink