nrf-rs/nrf-hal #113: Consider renaming repo name

FrozenDroid, 2019-09-14:
Currently there's nrf52 and nrf9160 crates in this repo, I think it'd be better if that was split up, or if the repo name reflected the contents of the repo. Maybe a generic "nrf-hal"?

Yatekii, 2019-09-14:
Thanks for raising the issue.
We are already aware and discussing internally to find the best solution.
Hopefully we can resolve this soon.

jonas-schievink, 2019-11-28:
We should try merging the nrf51-hal into this repo too, then naming it nrf5x-hal would be perfect. I'll also merge the nrf5340-hal soon.

thejpster, 2019-11-28:
> 
> 
> We should try merging the nrf51-hal into this repo too, then naming it nrf5x-hal would be perfect. I'll also merge the nrf5340-hal soon.

*cough* 9160 *cough*

jonas-schievink, 2019-11-28:
Err, yeah. https://github.com/nrf-rs/nrf52-hal/issues/125 proposes `nrf-hal`, I guess that's fine too.

Yatekii, 2019-11-28:
Yeah, I think we also discussed somewhere that we might leave the nrf51 support out of the general nrf-hal we are trying to create. Since it's a only thumv6 etc. But since there is now a thumbv8 one in the nrf53, I am not sure about that anymore. How close are the peripherals of the nRF51 and the nRF52?

jonas-schievink, 2019-11-28:
Some are pretty close (`TEMP`, `RADIO`), but other aren't. I haven't yet looked deeply into this though.

Yatekii, 2019-11-28:
I think the key point is that we don't sacrifice useability (many cfg flags or suboptimal API for some chips, etc.) just to merge them all, whatever it takes ;)

PR #139 (2020-04-11): Rename packages and repository
Closes #113.
closed 2020-04-11 by jonas-schievink