nrf-rs/nrf-hal #115: Provide periodic timer APIs

jonas-schievink, 2019-10-05:
The current timer APIs unconditionally enable the `compare0_stop` shortcut:

https://github.com/nrf-rs/nrf52-hal/blob/b2302446fb0317cd46a403c3eb36d5d7d1f69032/nrf52-hal-common/src/timer.rs#L37-L39

If this would be made configurable via a type parameter, `Timer` could provide a `Periodic` timer as well as the current one-shot implementation.

careyk007, 2019-10-15:
I'm interested in tackling this. Disabling the `compare0_stop` shortcut makes the timer periodic (as expected), but there are probably users who are expecting the default one-shot implementation. The options I have so far for exposing both approaches to the user are outlined below.

a) Pass an Enum to the `new` method. This would look something like this:
``` Rust
pub enum TimerType {
    OneShot,
    Periodic,
}

pub fn new(timer: T, type: TimerType) -> Self {
    match type {
        OneShot => {
            timer
                .shorts
                .write(|w| w.compare0_clear().enabled().compare0_stop().enabled());
        },
        Periodic => {
            timer
                .shorts
                .write(|w| w.compare0_clear().enabled().compare0_stop().disabled());
        }
    }
    // ...
}
```

b) Provide functions for starting a repeating timer and a one-shot timer:
``` Rust
pub fn start_oneshot(&mut self, cycles: u32) {
    self.0
        .shorts
        .write(|w| w.compare0_stop().enabled());

    self.start(cycles);
}

pub fn start_periodic(&mut self, cycles: u32) {
    self.0
        .shorts
        .write(|w| w.compare0_stop().disabled());

    self.start(cycles);
}
```

c) Use a state-machine approach similar to `embedded-hal`'s `into_push_pull_output` and `into_floating_input`, etc.:
``` Rust
pub fn into_oneshot(self) -> OneShotTimer {
    self.0
        .shorts
        .write(|w| w.compare0_stop().enabled());

    OneShotTimer(self.0)
}

pub fn into_periodic(self) -> PeriodicTimer {
    self.0
        .shorts
        .write(|w| w.compare0_stop().disabled());

    PeriodicTimer(self.0)
}
```

Option (a) would limit the use of the timer to a specific approach (`Periodic` or `OneShot`). I'm not sure how many users would be switching between a periodic and one-shot timer at runtime though, so it could be a valid approach. The major downside is that it would break backwards API compatibility.

Option (b) allows the timer to act as both a periodic and one-shot timer, configurable during runtime. The downside of this is that `timer.start()` is still exposed and will default to whichever approach was last used. Additionally, there is a small overhead added to each call to `start_periodic` and `start_oneshot`.

Option (c) seems like it might be the better approach, though I'll have to do some more work to flesh it out.

jonas-schievink, 2019-10-15:
Option (c) is the only way you can correctly implement the Periodic trait from the embedded-hal, so I recommend that approach.

I think it can be done backwards-compatibly by giving the timer a default type parameter too.

Yatekii, 2019-10-16:
People that really need both types of the same timer can easily use an enum to encapsulate it, so I think the third variant might be a good choice :)

jonas-schievink, 2019-10-18:
Closing since this was implemented in https://github.com/nrf-rs/nrf52-hal/pull/117

closed 2019-10-18 by jonas-schievink