nrf-rs/nrf-hal #120: Add gpiote module to nrf52-common-hal

chris-ricketts, 2019-11-10:
It would be nice to have a safe, ergonomic API to setup events and tasks using the GPIOTE module.

I have an [example](https://github.com/chris-ricketts/nrf52dk-rtfm-gpios/blob/master/src/main.rs) of how the PAC can be used directly, tested out on the nrf52 dev kit. 

I'm up for giving this a go.

I'm guessing it's going to be kind of similar to the `gpio` module but with more layers. Also I'm wondering about the best way to handle the multiple channels.

It definitely needs a typestate to indicate whether an event pin is configured as pullup or pulldown as a floating pin had me scratching my head for a while.

Does anyone have any hints or implementations they know of in other HAL crates that deal with similar functionality?

hannobraun, 2019-11-19:
Another usage example: https://github.com/braun-embedded/rust-dw1000/blob/34df744e4d0b83602191f7a48dfd4951ec143655/dwm1001/src/lib.rs#L884

jamesmunns, 2019-11-23:
@chris-ricketts have you made any progress on this so far?

chris-ricketts, 2019-11-25:
@jamesmunns I haven't got round to starting this yet - would you like to take it on?

kalkyl, 2020-06-26:
Has anyone started on this yet?
I've been using the gpiote quite a lot and have written a module to help with that. Should i open a PR with what i have where we could discuss and agree on an API?

jonas-schievink, 2020-06-26:
@kalkyl Go ahead! I always find it easier to discuss working code rather than ideas.

kalkyl, 2020-07-02:
PR #167 
Example:
https://github.com/kalkyl/nrf-hal/blob/gpiote/examples/gpiote-demo/src/main.rs

jonas-schievink, 2020-07-11:
https://github.com/nrf-rs/nrf-hal/pull/167 was merged.

closed 2020-07-11 by jonas-schievink