nrf-rs/nrf-hal #137: No 'new' method in gpio::p0::Parts

ade-sede, 2020-04-03:
I am poking around a nRF52840-DK.
I'm trying to implement a blinky for starters, but run into a puzzling compilation error.

```rustc
error[E0599]: no function or associated item named `new` found for struct `nrf52_hal_common::gpio::p0::Parts` in the current scope
  --> examples/blinky.rs:19:28
   |
19 |     let port0 = p0::Parts::new(peripherals.P0);
   |                            ^^^ function or associated item not found in `nrf52_hal_common::gpio::p0::Parts`

error: aborting due to previous error

For more information about this error, try `rustc --explain E0599`.
error: could not compile `nrf52840-dk`.

To learn more, run the command again with --verbose.
```

Source code: 
```rust
#![no_main]
#![no_std]

use cortex_m_rt::entry;
use cortex_m_semihosting::hprintln;

#[allow(unused_imports)]
use panic_semihosting;

use nrf52840_hal as hal;
use hal::nrf52840_pac as pac;
use hal::gpio::p0;

#[entry]
fn main() -> ! {
	hprintln!("Hello, world!").unwrap();

	let peripherals = pac::Peripherals::take().unwrap();
    let port0 = p0::Parts::new(peripherals.P0);

	loop {}
}
```

Cargo.toml (packaged info omitted):
```toml
[dependencies]
cortex-m = { version = "0.5.4", features = [ "const-fn" ] }
cortex-m-rt = "0.6.12"
embedded-hal = "0.2.3"
nrf52840-hal = "0.8.1"

[dev-dependencies]
cortex-m-rt = "0.6.12"
cortex-m-semihosting = "~0.3"
panic-semihosting = "~0.5"
nb = "~0.1"

[features]
rt = ["nrf52840-hal/rt"]
default = ["rt"]
```

I based this on the [SPI example](https://github.com/nrf-rs/nrf52-hal/blob/master/examples/spi-demo/src/main.rs).
I took a very quick look a the [gpio.rs](https://github.com/nrf-rs/nrf52-hal/blob/master/nrf52-hal-common/src/gpio.rs) file and if I correctly read the macro at the bottom, there should be a public method `new` implemented for `p0::Parts`

On the other hand, when looking at the HAL's [documentation](https://docs.rs/nrf52840-hal/0.8.1/nrf52840_hal/index.html) on doc.rs there is no `new` method implemented for `p0::Parts`.

From my naive point of view one of the following is happening:
- I am completely missing something (more than probable, as I am new to rust and even more to embedded rust)
- There is an error in the documentation (unlikely as it is automatically generated)
- There is an error in the example / the example is not portable to nRF52840
- An implementation mistake somehow

Can anyone help me figure that one out ?

jonas-schievink, 2020-04-03:
That's because the version on crates.io is fairly outdated. I hope we'll manage to release a new version soon :tm:.

ade-sede, 2020-04-03:
I see.
I take it should have no issue compiling this if I pull from here rather than crates.io
Thank you very much for this very quick answer, and for the work you guys are doing !

closed 2020-04-03 by ade-sede