nrf-rs/nrf-hal #150: Make sure all embedded-hal interfaces can implicitly copy

jamesmunns, 2020-05-25:
For compat reasons, we should support implicit copy (if the user passes in a slice from `.text`) of chunks on all interfaces. We currently do this for UARTE, but not for SPIM/TWIM.

kalkyl, 2020-06-23:
Instead of returning Error::DMABufferNotInDataMemory?

Yatekii, 2020-06-23:
I think the majority vote was in favor of returning an Error on the main interface but providing a call which does implicit copy if you ask it to.
But the interfaces should all have the same behavior/look :)

jamesmunns, 2020-06-23:
The goal was:

1. When using `embedded-hal` traits, implicitly copy if necessary (when we would otherwise return `DMABufferNotInDataMemory`)
2. When using `nrf-hal` direct interfaces, do not implicitly copy, instead return the error

This is because some (non nrf specific) drivers send constant strings, which will end up in `.text`.

kalkyl, 2020-06-24:
I had a got at it in PR #163 for spim transfer, if i got it right :)

kalkyl, 2020-06-24:
Oups, i messed up my first PR and accidently pushed the wrong branch, still getting used to the workflow of contributing, haha! Anyways, i now submitted the one i was supposted to (PR #164) for spim transfer :)

kalkyl, 2020-06-24:
Added Twim write/write_read aswell and made a new PR #165 with both Spim and Twim instead, sorry for the PR mess, i'm new here... :)

(Not sure why the githud build failed (##[error]Process completed with exit code 1), is that a common issue or is it me? It builds just fine on my machine...)