nrf-rs/nrf-hal #161: unresolved import `hal` in ecb-demo

jpwright, 2020-06-17:
Hi, I'm pretty new to Rust so apologies if this is something obvious on my end. When cloning the repo and running `cargo build`, I get the error shown below. I am using Ubuntu 19.10 and rustc version 1.44.0.

```
jason@faraday:/media/jason/e/projects/nrf-hal$ cargo build
   Compiling cortex-m v0.6.2
   Compiling typenum v1.12.0
   Compiling semver-parser v0.7.0
   Compiling stable_deref_trait v1.1.1
   Compiling vcell v0.1.2
   Compiling proc-macro2 v1.0.18
   Compiling unicode-xid v0.2.0
   Compiling syn v1.0.31
   Compiling cortex-m-rt v0.6.12
   Compiling r0 v0.2.2
   Compiling void v1.0.2
   Compiling nb v0.1.2
   Compiling nrf52832-pac v0.9.0
   Compiling nrf52833-pac v0.9.0
   Compiling nrf9160-pac v0.2.1
   Compiling nrf51 v0.9.0
   Compiling nrf52840-pac v0.9.0
   Compiling fpa v0.1.0
   Compiling nrf52810-pac v0.9.0
   Compiling rand_core v0.5.1
   Compiling memchr v2.3.3
   Compiling proc-macro2 v0.4.30
   Compiling nrf52832-hal v0.10.0 (/media/jason/e/projects/nrf-hal/nrf52832-hal)
   Compiling unicode-xid v0.1.0
   Compiling version_check v0.9.2
   Compiling version_check v0.1.5
   Compiling rand_core v0.4.2
   Compiling syn v0.15.44
   Compiling byteorder v1.3.4
   Compiling cortex-m-semihosting v0.3.5
   Compiling nrf9160-hal v0.10.0 (/media/jason/e/projects/nrf-hal/nrf9160-hal)
   Compiling heapless v0.4.4
   Compiling cortex-m v0.5.10
   Compiling panic-halt v0.2.0
   Compiling aligned v0.2.0
   Compiling ufmt-write v0.1.0
   Compiling ssd1306 v0.2.6
   Compiling cortex-m-rtfm v0.4.3
   Compiling nrf52840-hal v0.10.0 (/media/jason/e/projects/nrf-hal/nrf52840-hal)
   Compiling nrf51-hal v0.10.0 (/media/jason/e/projects/nrf-hal/nrf51-hal)
   Compiling adafruit-nrf52-bluefruit-le v0.0.1 (/media/jason/e/projects/nrf-hal/boards/adafruit-nrf52-bluefruit-le)
   Compiling nrf52833-hal v0.10.0 (/media/jason/e/projects/nrf-hal/nrf52833-hal)
   Compiling nrf52810-hal v0.10.0 (/media/jason/e/projects/nrf-hal/nrf52810-hal)
   Compiling volatile-register v0.2.0
   Compiling semver v0.9.0
   Compiling embedded-hal v0.2.3
   Compiling rand_core v0.3.1
   Compiling nom v5.1.2
   Compiling nom v4.2.3
   Compiling embedded-hal-spy v0.0.3
   Compiling rand v0.5.6
   Compiling rustc_version v0.2.3
   Compiling hash32 v0.1.1
   Compiling quote v1.0.7
   Compiling quote v0.6.13
   Compiling bare-metal v0.2.5
   Compiling cast v0.2.3
   Compiling tinytga v0.1.0
   Compiling generic-array v0.13.2
   Compiling generic-array v0.12.3
   Compiling generic-array v0.11.1
   Compiling as-slice v0.1.3
   Compiling aligned v0.3.2
   Compiling rtt-target v0.2.0
   Compiling panic-semihosting v0.5.3
   Compiling tinybmp v0.1.1
   Compiling embedded-graphics v0.4.9
   Compiling owned-singleton-macros v0.1.0
   Compiling cortex-m-rtfm-macros v0.4.3
   Compiling cortex-m-rt-macros v0.1.8
   Compiling owned-singleton v0.1.0
   Compiling ecb-demo v0.0.1 (/media/jason/e/projects/nrf-hal/examples/ecb-demo)
error[E0432]: unresolved import `hal`
  --> examples/ecb-demo/src/main.rs:22:5
   |
22 |     hal::{Clocks, Ecb},
   |     ^^^ use of undeclared type or module `hal`

error[E0433]: failed to resolve: use of undeclared type or module `hal`
  --> examples/ecb-demo/src/main.rs:34:13
   |
34 |     let p = hal::pac::Peripherals::take().unwrap();
   |             ^^^ use of undeclared type or module `hal`

error: aborting due to 2 previous errors

Some errors have detailed explanations: E0432, E0433.
For more information about an error, try `rustc --explain E0432`.
error: could not compile `ecb-demo`.

To learn more, run the command again with --verbose.
warning: build failed, waiting for other jobs to finish...
error: build failed
```

thalesfragoso, 2020-06-17:
The demo is configured to be able to build for the different microcontrollers, so you need to specify which one you're using through cargo features. You might also want to specify your target platform, see the [readme](https://github.com/nrf-rs/nrf-hal/blob/master/examples/ecb-demo/README.md) on the example for more details.

jonas-schievink, 2020-06-17:
The error message here could be improved by using `compile_error!`