nrf-rs/nrf-hal #182: Question: I try use set_high() method but not found

iddar, 2020-07-18:
I try to compile mi first blink for nrf52840 uC and review your examples for learning and see use directly `set_low()` after creating Pin instance  with `into_push_pull_output`

follow this example: https://github.com/nrf-rs/nrf-hal/blob/master/examples/spi-demo/src/main.rs#L31

this is my code
```rust
#![no_std]
#![no_main]

extern crate panic_halt;
use cortex_m_rt::entry;

extern crate nrf52840_hal;
use nrf52840_hal::gpio;
use nrf52840_hal::gpio::*;
use nrf52840_hal::gpio::p0::*;

use nrf52840_hal::Timer;

#[entry]
fn main() -> ! {
    let p = nrf52840_hal::pac::Peripherals::take().unwrap();
    let p0 = p0::Parts::new(p.P0);
    let mut led1: P0_24<gpio::Output<PushPull>> = p0.p0_24.into_push_pull_output(Level::High);

    let mut timer = Timer::new(p.TIMER0);

    loop {
        led1.set_low();
        timer.delay(2_000_000);
        led1.set_high();
        timer.delay(2_000_000);
    }
}
```

And this is an error message, both equal for `set_low` and `set_high`

```
error[E0599]: no method named `set_low` found for struct `nrf52840_hal::gpio::p0::P0_24<nrf52840_hal::gpio::Output<nrf52840_hal::gpio::PushPull>>` in the current scope
  --> src/main.rs:26:14
   |
26 |         led1.set_low();
   |              ^^^^^^^ method not found in `nrf52840_hal::gpio::p0::P0_24<nrf52840_hal::gpio::Output<nrf52840_hal::gpio::PushPull>>`

```

thalesfragoso, 2020-07-18:
`set_low` and `set_high` are from traits defined in `embededd-hal`, the traits must be in scope for them to work.

In the example you posted it was in this line
```rust
use embedded_hal::digital::v2::OutputPin;
``` 

but I think you will be better doing:
```rust
use nrf52840_hal::prelude::*;
```

iddar, 2020-07-18:
Thanks for your quick response, my goal is to be able to add a new plate that my friends created.

I will continue learning from your catch and your experience, thank you

closed 2020-07-18 by iddar