nrf-rs/nrf-hal #205: nRF52840 Temp needs to be divided by 4?

daschl, 2020-08-29:
Hi,

I've been toying around with the hal for the nRF52840 and reading the internal die temperature sensor. Something like this:

```rs
    let peripherals = Peripherals::take().unwrap();

    let mut die_temp_sensor = Temp::new(peripherals.TEMP);

    let die_temp_c = die_temp_sensor.measure().to_bits() / 4;
    defmt::info!("die temp is {:i32}°C", die_temp_c);
```

Now after much digging I found out that while (as the rust docs say)  the temp is in celsius, it actually needs to be divided by 4 (I think this is because it's 0.25c steps? https://infocenter.nordicsemi.com/index.jsp?topic=%2Fps_nrf52840%2Ftemp.html&cp=3_0_0_5_27)

So I wonder if we should either

 - divide it by 4 in the `measure()` method (or the one it calls) directly
 - or at least add something to the rustdoc indicating it needs to be divided by 4?

In any case, I'm happy to work on a PR whatever gets decided by the authors. Thanks for the project, it's great and fun to work with!

jonas-schievink, 2020-08-29:
That's why `measure` returns a fixed-point number with 2 bits past the point, but you're converting it to its raw bits by calling `.to_bits()` on it.

daschl, 2020-08-29:
@jonas-schievink can you help me understand how that relates to the multiple of 4? Is there a different way to get the correct celsius value into an integer?

jonas-schievink, 2020-08-29:
The fixed-point type already does the division by 4. You undid it by calling `.to_bits()`. On master we've switched to the `fixed` crate for those, you can see the API here: https://docs.rs/fixed/1.1.0/fixed/struct.FixedI32.html

daschl, 2020-08-30:
@jonas-schievink thanks for the reply, one thing I still couldn't figure out: if `.to_bits()` is undoing it, how can I turn the fixed i32 into the primitive "correct" i32 equivalent? For example I'm trying to print the number with defmt but I couldn't get it to work so I used `to_bits()`. I mean I'm okay to just divide it by 4 again on my own, but based on your comment that doesn't seem the wisest thing to do.

jonas-schievink, 2020-08-30:
`FixedI32` implements various `Cast*` traits, you can probably use those to convert to an integer.

daschl, 2020-08-30:
Ok, so in case someone is running into the same issue, I got it working by calling `to_num()` on it, as in

```
let die_temp_c: i32 = die_temp_sensor.measure().to_num();
defmt::info!("die temp is {:i32}°C", die_temp_c);
```

@jonas-schievink do you think it could make sense to improve the rustdoc pointing users to the "good" method to use? I'd be happy to try my hands on that improvement.

jonas-schievink, 2020-08-30:
I'd be happy to merge a doc example showing this (provided that it is at least build-tested in CI)

daschl, 2020-09-06:
Okay thanks, I'll close this for now and follow up with a PR

closed 2020-09-06 by daschl