nrf-rs/nrf-hal #213: gpio port not actually Floating

daschl, 2020-09-06:
Hi,

I'm not sure if this is expected, but it at least looks "weird" to me. The issue is that my Input, even though marked as Floating, doesn't work out of the box, only if I call `into_floating_input` explicitly on it.

So, consider this code:

```rs
let peripherals = Peripherals::take().unwrap();
let port1 = p1::Parts::new(peripherals.P1);
let external_temp_sensor = Tsic::new(port1.p1_08.into_floating_input().degrade());
```

This work as expected (my `Tsic` takes `Pin<Input<Floating>>` as an argument). But if I do it like this, it compiles, but I'll never get the high and low states of the pin:

```rs
let peripherals = Peripherals::take().unwrap();
let port1 = p1::Parts::new(peripherals.P1);
let external_temp_sensor = Tsic::new(port1.p1_08.degrade());
```

Would it be possible that the port on the peripheral is neither `Floating`, nor `PullDown` or `PullUp` but it would always have to be turned `into` one of these types? This would make it explicit that it needs to be called, I couldn't figure out why my sensor didn't work and only stumbled on this by trial and error. Or is there something else going on that I'm missing?

jonas-schievink, 2020-09-06:
Hmm, what does the datasheet say? Maybe not every pin starts in floating input mode?

daschl, 2020-09-06:
@jonas-schievink I couldn't locate it in the datasheet. Sorry forgot to mention it's the 52840 DK and the pin I use is P1.08

jonas-schievink, 2020-09-11:
Oh, I see the problem. The `PIN_CNF` registers are all initialized with `INPUT` set to `Disconnect`, so whatever is on the pin doesn't actually get sampled into the input buffer. Calling `into_floating_input` explicitly connects the input buffer.

jonas-schievink, 2020-09-11:
This seems to be the case for *every* nRF chip. I wonder why nobody hit this before.

I think the proper fix would be to introduce a new "Disconnected" or "Reset" pin state and make every pin start in this state.

jonas-schievink, 2020-09-11:
@daschl Can you try out https://github.com/nrf-rs/nrf-hal/pull/220? That should fix this issue.

daschl, 2020-09-11:
I won’t be able to test it over the weekend, but next week I’ll give it a try. From looking at the change it makes a lot of sense to me.

closed 2020-09-12 by jamesmunns

daschl, 2020-09-14:
Got it tested, works great - thanks!

PR #220 (2020-09-12): Make GPIOs start in a `Disconnected` state
This should fix #213 by making all GPIO pins start in a new `Disconnected` state that does not implement any of the embedded-hal traits and needs to be converted to a functional state first.

I've also implemented `Channel<Saadc>` for *all* GPIO type states (even output states), which should be correct according to the GPIO schematics in the manual. Using a pin as a SAADC channel will fully switch the pin from the GPIO circuitry to the analog circuitry. This avoids breaking code that passes an uninitialized pin to the SAADC.