nrf-rs/nrf-hal #222: Usage with softdevice

kuon, 2020-09-12:
I'd like to write my nrf52840 firmware in rust, but I need stable bluetooth (I found rubble, which I got working with this crate, but it seems very alpha in development). For this, I'd like to use soft device s140, with a bit of C to use the softdevice (and avoid exposing all softdevice features to rust).

I can use this crate without issue without the soft device, but I am wondering if it is possible to use this crate with a soft device.

I tried different linker script and flags, but I can't make it work.

I put my test here:

https://github.com/kuon/nrf_s140/blob/master/nrf.ld

I inspired my `.cargo/config` from https://github.com/jamesmunns/nrf52dk-sys/blob/master/.cargo/config but I don't think the settings are picked by crate users (I had to copy it in my binary crate).

Dirbaio, 2020-09-12:
Using softdevice + nrf-hal is definitely possible.

Going the "wrapping Nordic's C SDK code" route is quite painful though. I tried that first and kind-of gave up. Here's some of what you'll encounter:
- Rust controls the interrupt vector, you'll have to make sure to forward whatever interrupts the C SDK code needs from Rust to C.
- The SoftDevice needs exclusive access to some peripherals (RADIO, some TIMERs, SWI/EGUs, ECB, AAR..), you have to make sure not to use them on your own.
- The SoftDevice needs exclusive access to some IRQs and priority levels - you have to make sure Rust code doesn't use them. Nordic gives you some "safe" NVIC wrappers at `s140/headers/nrf_nvic.h` that check that for you, but if you touch interrupts or critical sections from Rust code you're bypassing that. It has some globals (`nrf_nvic_state`) that you have to make sure to keep in sync between C and Rust code.
- You'll have to make sure you don't use nrf_hal and nrfx in ways that conflict fighting over the same peripherals.

For these reasons, instead of wrapping the SDK C code, I've started working on some pure-Rust crates to safely use the softdevice together with nrf-pac and nrf-hal: https://github.com/akiles/nrf-softdevice. 

- `nrf-softdevice-sXXX` crates are the raw (unsafe) softdevice bindings generated with bindgen
- `nrf-softdevice` wraps the bindings with safe idiomatic Rust APIs. The idea is to recreate in Rust the very few parts of the C SDK code that are needed to use the softdevice. Safe interrupt and flash APIs are complete, and bluetooth is WIP but usable already (help would be welcome!)

There are some [examples](https://github.com/akiles/nrf-softdevice/tree/master/example/src/bin) of s140+nrf52840, you can use nrf-hal with these as-is and it should just work.

kuon, 2020-09-12:
I didn't think of using soft device without C, that's surely quite some work, but I guess it is worth it. The Nordic SDK is quite hard to navigate, so I'd be happy to be able to make it work without it.

I will look at your work in details and do my best to contribute.

I have a few projects where I want to use the nrf52840 so I should have quite some time to contribute.

-- 
Nicolas Goy

closed 2020-09-12 by kuon