nrf-rs/nrf-hal #249: UART0 vs UARTE0 - serial to USB on Adafruit ItsyBitsy nRF52840

nebelgrau77, 2020-10-13:
I managed to put together some code that prints values from ADC to serial on the Adafruit ItsyBitsy nRF52840 Express. It all works perfectly with UARTE. The only thing is, I need to use an FTDI-USB board to read serial. It seems that the UART pins on this board D0 = P0_25 for TX and D1=P0_24 for RX) should be connected to USB. E.g, the TinyGo package implements that: https://tinygo.org/microcontrollers/machine/itsybitsy-nrf52840/ - so the first question is, would I need UART0 instead of UARTE0 for that to work? At the moment my computer can see the serial device on ttyUSB0, but cannot access it. The nRF52840-HAL crate does not implement Uart, only Uarte. I tried adding it in my fork, but it won't really work. Am I barking at the wrong tree here and the problem is something else?

nebelgrau77, 2020-10-13:
It seems like there is no actual connection between USB and the serial on the board, seems like it's all done in software, e.g. when you use the board with CircuitPython. I will leave it open, though, in case someone has some better insight.

Dirbaio, 2020-10-19:
"UART" is the legacy peripheral compatible with nRF51 series. "UARTE" is the newer peripheral Norfic made for the nRF52 series, which supports tasks, events and DMA. UARTE can do everything UART

nebelgrau77, 2020-10-20:
OK, gotcha. So basically I can only use UARTE on nRF52, as UART is not implemented, and it will do everything UART0 does. 

I don't even know if there is any problem in the first place: my code works fine as is, both ways. I need an FTDI-USB connected to TX/RX pins in the middle, though. I was under impression (and this is caused by the description of the corresponding package for TinyGo) that this particular board also had the UART0 connected to the USB, i.e. I wouldn't need the FTDI-USB in the middle. This does not seem to  be confirmed by the schematics, at least from my limited understanding of them.  

So I assume that the serial-over-USB on this board does somehow work on software/bootloader level: the Circuit Python section of the board instruction does in fact mention using the TX and RX pins with FTDI-USB as "hardware serial", while the terminal access to Circuit Python is over USB.

Dirbaio, 2020-10-20:
Yes, the USB on the board goes directly to the nrf52's D+ and D- pins as you can see on the schematics. The bootloader speaks USB directly and makes the board appear to the PC as an USB serial device, there's no USB-to-serial hardware.

You could do the same in your firmware with `usb-device` and `usbd-serial` crates but [it's not ready for use yet](https://github.com/nrf-rs/nrf-hal/pull/199) :(

nebelgrau77, 2020-10-20:
OK, clear. As I said, it's not an issue (not for me anyway). I was just curious whether I was missing something, and the Adafruit website doesn't go into detail on this. Thanks again, I will close the issue at this point.

closed 2020-10-20 by nebelgrau77