nrf-rs/nrf-hal #251: error compile w7

tcpipchip, 2020-10-19:
Hi

is that a windows 7 problem only ?

[image]

Can you help me  ?

jonas-schievink, 2020-10-19:
`cargo-embed` is a host application, so it cannot be built for microcontroller targets like `thumbv7em-none-eabihf`. It also isn't part of the nRF HALs, but probe-rs (https://github.com/probe-rs/cargo-embed).

This looks like a misconfiguration of your system, since it is trying to compile cargo-embed for `thumbv7em-none-eabihf`.

tcpipchip, 2020-10-19:
@jonas-schievink 
Thanks to answer fast
The true is that i want to compile this
https://github.com/nrf-rs/nrf-hal/tree/master/examples/blinky-button-demo 
to the NRF52832/NRF52840
Can you instruct me ?

tcpipchip, 2020-10-19:
this ?
[image]

tcpipchip, 2020-10-19:
success compile!

Help me about the error that i got after try to install probe-run
[image]

jonas-schievink, 2020-10-19:
I don't know, but it's not an nrf-hal problem, so closing this issue. With general Rust questions like this, I recommend asking on https://users.rust-lang.org/.

closed 2020-10-19 by jonas-schievink