nrf-rs/nrf-hal #262: Consider adding more support for the nRF9160

macthestack, 2020-11-06:
Many modules are turned off in the nrf-hal-common crate like for example gpiote. The examples are also geared towards 51 and 52.

Is someone working on this? I could give it a try myself but I'm new to embedded programming.

Yatekii, 2020-11-06:
Hi!

We are constantly adding features to the HALs, but of course, pretty much everyone does this in their sparetime, so this is sometimes going slower and sometimes going faster :) I believe not many folks are working with the nRF91 so it's not so well supported.

It would be awesome if you had time & motivation to add more features for it :) Being new to embedded programming is no issue, we love to provide feedback and help where needed :)

You could also join our matrix channel https://matrix.to/#/!dASuPAbUCerhCYFqJs:matrix.org?via=matrix.org for more interactive help :)

macthestack, 2020-11-12:
Thank you for guiding me to the matrix channel and the breakthrough help I received there. Hopefully, I can get to a stage when I can share my work, soon :)

closed 2020-11-12 by macthestack