nrf-rs/nrf-hal #263: nRF52840 HAL - importing from common HAL

nebelgrau77, 2020-11-07:
How do I import modules from the common HAL crate? 

This works fine:

```rust
use nrf52840_hal as hal;
use hal::rng;
```
 
but this doesn't, 

```rust
use nrf52840_hal as hal;
use hal::qdec::*;
```

error: ^^^ no qdec in the root

Same for i2s. Looking at how various modules are exposed in lib.rs, I don't understand why rng yes, but qdec no. This is how the HAL is defined in Cargo.toml:

```
nrf52840-hal = { version = "*", features = ["rt"]}
```

What am I doing wrong here?

Yatekii, 2020-11-07:
Hi :)

Those two modules have only recently been added thanks to @kalkyl's untiring work :)
A release is iminent; until then you could use master :)

nebelgrau77, 2020-11-08:
Thanks, that makes sense! 

I stumbled upon another glitch, I will open it as a separate issue.

closed 2020-11-08 by nebelgrau77