nrf-rs/nrf-hal #264: nb crate version conflict between 0.11.1 and embedded-hal

nebelgrau77, 2020-11-08:
I just stumbled upon a problem that seems to be related to this. I'm using the master branch, and here's what happens:

Bottomline: nrf-hal-common won't compile. The reason is this:

```rust
 = note: expected fn pointer `fn(&mut saadc::Saadc, &mut PIN) -> core::result::Result<_, nb::Error<()>>` (enum `nb::Error`)
               found fn pointer `fn(&mut saadc::Saadc, &mut PIN) -> core::result::Result<_, nb::Error<()>>` (enum `nb::Error`)
 = note: perhaps two different versions of crate `nb` are being used?
```

This affects `saadc` and `timer` modules. 
In two cases the "expected" and "found" look identical as seen above, but then they aren't:

```rust
    = note: expected enum `nb::Error<void::Void>`
               found enum `nb::Error<_>`
    = note: perhaps two different versions of crate `nb` are being used?
```

If I run `cargo tree` it does confirm that two different versions of nb are used: v1.0.0 in the nrf-common-hal, and v0.1.2 in embedded-hal. 

So I checked something that works for me: https://github.com/nebelgrau77/ItsyBitsy_nRF52840_ADC_serialTX - this uses the v0.11.0 of the nrf-common-hal, pulled from crates.io, so the nb version was still v0.1.2. 

In fact downgrading it back to 0.1.2 in a local fork of the master fixed it. 

There's a big fat chance I'm missing something, though! :)

_Originally posted by @nebelgrau77 in https://github.com/nrf-rs/nrf-hal/pull/179#issuecomment-723559835_

eldruin, 2020-11-09:
Have you tried running `cargo update`? `nb` `0.1.3` is compatible with `nb` `1.0.0`

nebelgrau77, 2020-11-09:
That solved the problem, thank you!

closed 2020-11-09 by nebelgrau77