nrf-rs/nrf-hal #270: Example gpiote-demo only works when debugging on nRF52840-DK

jamolnng, 2020-11-28:
OS: Windows 10
Rust version:  1.48.0
Target: thumbv7em-none-eabihf
Platform: nRF52840-DK (10056)

This is a weird one I've been trying to figure out all night. I cannot get the gpiote-demo to work when running the debug or release binary except when I am in debugging mode.

I am using an exact copy of the code/Cargo.toml/Embed.toml from the example. The only thing that works is btn2 toggling the LED1. Buttons 1, 3, and 4 do nothing when not debugging.

When debugging, buttons 1, 3, and 4 behave as intended and this is seen with the `rprintln!` statements appearing, when I set break points in `on_gpiote` and `debounce` and they get called, or when I see the LEDs changing as intended.

When I try to manually toggle the LEDs in `on_gpiote` or `debounce` when not debugging nothing happens, which is leading me to believe that `on_gpiote` and `debounce` are not getting triggered when the processor is not being debugged, which seems strange.

Finally, when not debugging I changed which button uses the PPI to toggle the LED and each button works when they use that.

So I'm really not sure what's going on. The only two things I can think is that it might, but unlikely, be an issue with the ext hfosc since that is bypassed in debugging mode and the debugger's clock is used, but I've tested without it enabled and still have the same issues. Or it also be an issue with this board but I'm not sure how likely that is.

Code here: https://gist.github.com/jamolnng/54d65593488609ae3149eaa665182cb9

jamolnng, 2020-11-29:
This just keeps getting weirder to me. When I change line 100 in main.rs to this
```rust
        ctx.schedule.debounce(ctx.start + 5000.cycles()).ok();
```
it works as "intended" when i press buttons 1, 3, and 4 from 2-5 times first with the most common of button presses required to be 4. If I go up to `5100.cycles()` or down to `4900.cycles()` it goes back to not working at all

jamolnng, 2020-11-29:
https://github.com/rtic-rs/cortex-m-rtic/issues/331

seems this is actually an issue with rtic

thvdveld, 2024-07-04:
I'm also hitting this issue with GPIOTE and RTIC not working. @jamolnng did you find the issue or a possible fix?