nrf-rs/nrf-hal #275: Testing

huntc, 2020-12-27:
Firstly, what an amazing effort! Thank you for this initiative!

Being new to the embedded world, I note a severe lack of unit tests. As C is often used, I put that down to the effort of setting up a test framework being a deterrent. With Rust, unit testing is easy, and my own embedded development in Rust has been able to incorporate unit tests. 

I realise that stubbing out peripherals can be particularly tricky, but I’m curious to learn more about the testing strategies used here. I’m also keen to learn whether unit tests could be applied to this repo. Thanks.

Dirbaio, 2021-01-11:
IMO tests that mock register reads/writes have limited value, wrt the required effort.

Many bugs in peripheral drivers are due to the peripheral not doing what you think it does given some register writes. (due to quirky behavior, ambiguous docs...). Testing that the code does the expected register writes doesn't really test that the hardware does what it should do.

jonas-schievink, 2021-01-11:
If you want a *good* example of a well-tested embedded driver, check out [bxcan](https://github.com/stm32-rs/bxcan).

CAN is simple to test because there is premade hardware where only the 2 CAN buses need to be connected to get something that can, in principle, test the entire peripheral. This is not the case for most other peripherals, so most HALs don't bother doing this.

If there was more industrial interest in embedded Rust, instead of almost all HALs being written by hobbyists, building such a testsuite to ensure reliability would make more sense.