nrf-rs/nrf-hal #288: Examples do not flash out of tree

mfiumara, 2021-01-25:
When copy pasting the example code outside of the source tree and correclty changing the nrf-hal path, embed does not flash the target binary correctly to the device:
```
       Error failed to flash /home/mattia/experiments/nrf-rust-workspace/qdec-demo/target/thumbv7em-none-eabihf/debug/qdec-demo
             
             Caused by:
                 No loadable ELF sections were found.
```
Cargo.toml:
```
"../nrf-hal/nrf52840-hal"
```
Workspace:
```
./nrf-hal
./qdec-demo
```

jonas-schievink, 2021-01-25:
Please refer to the directory layout in https://github.com/rust-embedded/cortex-m-quickstart, you need to configure Cargo correctly before targeting Cortex-M devices will work

closed 2021-01-25 by jonas-schievink

mfiumara, 2021-01-25:
Hi @jonas-schievink,

Sorry I'm quit the noob with regards to rust, I guess my error message indicates the link script can't be found.

Checking your link and comparing with the example code in this repo the only path that I see needs changing is `nrf52840-hal = { features = ["rt"], path = "../../nrf52840-hal" }` in `Cargo.toml`, is there some additional configuration which I'm missing? I'm building using `cargo embed --target thumbv7em-none-eabihf`.

I'm struggling to see why moving the example code up a folder would break flashing.

jonas-schievink, 2021-01-25:
You need a Cargo configuration file like this: https://github.com/nrf-rs/nrf-hal/blob/master/.cargo/config

mfiumara, 2021-01-25:
Ahhh that's what I was missing, thanks it seems to be working now. (Y)