nrf-rs/nrf-hal #296: Flash address for non-secure nRF9160 application

jonathanpallant, 2021-02-18:
We currently link nrf9160 applications at address 0x0004_0000. It seems Nordic have changed the address to be 0xc000 in their latest Secure Platform Manager.

```
SPM: NS image at 0xc000
SPM: NS MSP at 0x20028620
SPM: NS reset vector at 0xe7a9
SPM: prepare to jump to Non-Secure image.
*** Booting Zephyr OS build v2.4.0-ncs2  ***
The AT host sample started
```

We should probably update nrf9160-hal to use this address. I did try, but then I fell over a bunch of errors around pins being `Input` vs `Disconnected` which suggests nrf9160-dk doesn't build with the latest HAL.

```
   --> /home/user/nrf9160-dk/src/lib.rs:440:24
    |
440 |                 P0_01: pins0.p0_01,
    |                        ^^^^^^^^^^^ expected struct `Input`, found struct `Disconnected`
    |
    = note: expected struct `P0_01<Input<Floating>>`
               found struct `P0_01<Disconnected>`
```

Is there a working example using the latest SPM that I can crib from?

jonathanpallant, 2021-02-18:
I have checked out v0.10.0 and patched it to use flash address `0xC000` and I can confirm I can get firmware to boot. At some point I'll try and work out what's happened to the HAL to make my BSP no longer build.

huntc, 2021-05-04:
Interesting, [the Nordic doco](https://infocenter.nordicsemi.com/index.jsp?topic=%2Fps_nrf9160%2Fmemory.html) is still claiming that unsecured apps start at 0x0004_0000. [Looks like the MCUboot got fatter](https://github.com/nrfconnect/sdk-zephyr/commit/d5c822d5c299029d9b87dfdddf21f070a3910fd4).

Are you sure that the unsecure flash address is now 0xC000 though (that looks like the unsecured flash length of slot1 to me)? I *think* the unsecured flash address for apps is now lower at 0x0003_0000. I find the reg configuration interesting as it appears to be the size and then the address. [Here's the change I'm looking at](https://github.com/nrfconnect/sdk-zephyr/commit/d5c822d5c299029d9b87dfdddf21f070a3910fd4#diff-f13873893894ac51ebd55b082a63be406f8154007ddbd98135bf6df45987ce48R29).

huntc, 2021-05-04:
So, it seems that the unsecure flash address is 0x0005_0000. I've just managed to flash the sample along with Nordic's Secure Partition Manager (`west build -p auto -b nrf9160dk_nrf9160 ../nrf/samples/spm` from with the zephyr folder):

```
*** Booting Zephyr OS build v2.4.99-ncs1-3510-ge4f87fba1b87  ***
Flash regions           Domain          Permissions
00 09 0x00000 0x50000   Secure          rwxl
10 31 0x50000 0x100000  Non-Secure      rwxl

SRAM region             Domain          Permissions
00 07 0x00000 0x10000   Secure          rwxl
08 31 0x10000 0x40000   Non-Secure      rwxl

Peripheral              Domain          Status
00 NRF_P0               Non-Secure      OK
01 NRF_CLOCK            Non-Secure      OK
02 NRF_RTC0             Non-Secure      OK
03 NRF_RTC1             Non-Secure      OK
04 NRF_NVMC             Non-Secure      OK
05 NRF_UARTE1           Non-Secure      OK
06 NRF_UARTE2           Secure          SKIP
07 NRF_TWIM2            Non-Secure      OK
08 NRF_SPIM3            Non-Secure      OK
09 NRF_TIMER0           Non-Secure      OK
10 NRF_TIMER1           Non-Secure      OK
11 NRF_TIMER2           Non-Secure      OK
12 NRF_SAADC            Non-Secure      OK
13 NRF_PWM0             Non-Secure      OK
14 NRF_PWM1             Non-Secure      OK
15 NRF_PWM2             Non-Secure      OK
16 NRF_PWM3             Non-Secure      OK
17 NRF_WDT              Non-Secure      OK
18 NRF_IPC              Non-Secure      OK
19 NRF_VMC              Non-Secure      OK
20 NRF_FPU              Non-Secure      OK
21 NRF_EGU1             Non-Secure      OK
22 NRF_EGU2             Non-Secure      OK
23 NRF_DPPIC            Non-Secure      OK
24 NRF_REGULATORS       Non-Secure      OK
25 NRF_PDM              Non-Secure      OK
26 NRF_I2S              Non-Secure      OK
27 NRF_GPIOTE1          Non-Secure      OK

SPM: NS image at 0x50000
SPM: NS MSP at 0x20040000
SPM: NS reset vector at 0x50145
SPM: prepare to jump to Non-Secure image.
Hello, world!
```

jonathanpallant, 2021-05-04:
I have a feeling the partition layouts are dynamically calculated by `west` at build time, based on .... something. You can see in my terminal capture it definitely said `SPM: NS image at 0xc000`.

jonathanpallant, 2021-05-04:
Could you try building the AT host sample (one that says "The AT host sample started" on start-up) and see if the partition layout is different to when building spm by itself?

huntc, 2021-05-05:
I'm having difficulty building the nRF samples so I can't try flashing the AT client right now.

The partition layouts do indeed seem to be variable. I've added a link to the SPM I've used within the https://github.com/nrf-rs/nrf9160-dk/pull/2 PR. Meanwhile, I think we should close this particular issue as the 9160 hal's memory declaration is at least consistent with the Nordic documentation. WDYT?

huntc, 2021-05-05:
@jonathanpallant Furthermore, I've created a PR to update this bsp to the current hal: https://github.com/nrf-rs/nrf9160-dk/pull/4

jonathanpallant, 2021-05-14:
OK, so if I build spm by itself, I see:

```
*** Booting Zephyr OS build v2.4.99-ncs2  ***
Flash regions		Domain		Permissions
00 09 0x00000 0x50000 	Secure		rwxl
10 31 0x50000 0x100000 	Non-Secure	rwxl

SRAM region		Domain		Permissions
00 07 0x00000 0x10000 	Secure		rwxl
08 31 0x10000 0x40000 	Non-Secure	rwxl

Peripheral		Domain		Status
00 NRF_P0               Non-Secure	OK
01 NRF_CLOCK            Non-Secure	OK
02 NRF_RTC0             Non-Secure	OK
03 NRF_RTC1             Non-Secure	OK
04 NRF_NVMC             Non-Secure	OK
05 NRF_UARTE1           Non-Secure	OK
06 NRF_UARTE2           Secure		SKIP
07 NRF_TWIM2            Non-Secure	OK
08 NRF_SPIM3            Non-Secure	OK
09 NRF_TIMER0           Non-Secure	OK
10 NRF_TIMER1           Non-Secure	OK
11 NRF_TIMER2           Non-Secure	OK
12 NRF_SAADC            Non-Secure	OK
13 NRF_PWM0             Non-Secure	OK
14 NRF_PWM1             Non-Secure	OK
15 NRF_PWM2             Non-Secure	OK
16 NRF_PWM3             Non-Secure	OK
17 NRF_WDT              Non-Secure	OK
18 NRF_IPC              Non-Secure	OK
19 NRF_VMC              Non-Secure	OK
20 NRF_FPU              Non-Secure	OK
21 NRF_EGU1             Non-Secure	OK
22 NRF_EGU2             Non-Secure	OK
23 NRF_DPPIC            Non-Secure	OK
24 NRF_REGULATORS       Non-Secure	OK
25 NRF_GPIOTE1          Non-Secure	OK

SPM: NS image at 0x50000 <----------------------
SPM: NS MSP at 0x20040000
SPM: NS reset vector at 0x50145
SPM: prepare to jump to Non-Secure image.
```

If I build `at_sample` I see:

```
*** Booting Zephyr OS build v2.4.99-ncs1  ***
Flash regions		Domain		Permissions
00 00 0x00000 0x08000 	Secure		rwxl
01 31 0x08000 0x100000 	Non-Secure	rwxl

Non-secure callable region 0 placed in flash region 0 with size 32.

SRAM region		Domain		Permissions
00 07 0x00000 0x10000 	Secure		rwxl
08 31 0x10000 0x40000 	Non-Secure	rwxl

Peripheral		Domain		Status
00 NRF_P0               Non-Secure	OK
01 NRF_CLOCK            Non-Secure	OK
02 NRF_RTC0             Non-Secure	OK
03 NRF_RTC1             Non-Secure	OK
04 NRF_NVMC             Non-Secure	OK
05 NRF_UARTE1           Non-Secure	OK
06 NRF_UARTE2           Secure		SKIP
07 NRF_TWIM2            Non-Secure	OK
08 NRF_SPIM3            Non-Secure	OK
09 NRF_TIMER0           Non-Secure	OK
10 NRF_TIMER1           Non-Secure	OK
11 NRF_TIMER2           Non-Secure	OK
12 NRF_SAADC            Non-Secure	OK
13 NRF_PWM0             Non-Secure	OK
14 NRF_PWM1             Non-Secure	OK
15 NRF_PWM2             Non-Secure	OK
16 NRF_PWM3             Non-Secure	OK
17 NRF_WDT              Non-Secure	OK
18 NRF_IPC              Non-Secure	OK
19 NRF_VMC              Non-Secure	OK
20 NRF_FPU              Non-Secure	OK
21 NRF_EGU1             Non-Secure	OK
22 NRF_EGU2             Non-Secure	OK
23 NRF_DPPIC            Non-Secure	OK
24 NRF_REGULATORS       Non-Secure	OK
25 NRF_GPIOTE1          Non-Secure	OK

SPM: NS image at 0xc000 <----------------------
SPM: NS MSP at 0x2001c350
SPM: NS reset vector at 0xf171
SPM: prepare to jump to Non-Secure image.
```

This is using NCS v1.5.1.

I'm happy we can direct people to just compile `spm` and use that to get into NS mode. The  steps are:

```console
$ west init -m https://github.com/nrfconnect/sdk-nrf --mr v1.5.1 ncs_151
$ cd ncs_151
$ west update
$ # wait a very long time for all the `git clone` operations to complete
$ cd nrf/samples/spm
$ west build --board=nrf9160dk_nrf9160
$ west flash
```

jonathanpallant, 2021-05-14:
See https://github.com/nrf-rs/nrf-hal/pull/321

huntc, 2021-05-14:
Given https://github.com/nrf-rs/nrf-hal/pull/321, I think we can now close this. Wdyt?

closed 2021-05-17 by jonathanpallant