nrf-rs/nrf-hal #306: Q: Is it possible that nrf52820 is supported?

Skosulor, 2021-04-01:
Hi, thanks for this amazing repo!

I have a question regarding support for nrf52820, I can see in the list of supported devices that it is not supported however the nrf52833 is supported. Nordic-semiconductor recommends using the nrf52833 dev board for nrf52820 development[1]. Is it possible that compilations for the nrf52833 could work for the nrf52820? I would assume it requires some minor tweaks. 

[1] https://www.nordicsemi.com/Software-and-tools/Development-Kits/nRF52833-DK

Yatekii, 2021-04-01:
Yes, you can add some minor cfg tweaks to the code and copy & rename the nrf52833-hal directory and you should be good to go. We do this for all variants. Basically everything is in nrf-hal-common with cfg flags for different chips :)

Skosulor, 2021-04-06:
Awesome, Thanks! :)

closed 2021-04-06 by Skosulor