nrf-rs/nrf-hal #336: A common HAL NVMC abstraction - needed?

huntc, 2021-07-12:
Do we need a common-hal abstraction for the NVMC as per https://github.com/NordicSemiconductor/nrfx/blob/7eca6c2dc02b24cbdaa3ba0e63a7195e34ebe07c/hal/nrf_nvmc.h#L259

Dirbaio, 2021-07-13:
The driver could impl `NorFlash` and `MultiwriteNorFlash` from `embedded_storage` :)

closed 2021-07-20 by bors[bot]

ia0, 2022-01-08:
Hi,

AFAIU #337 did not implement `MultiwriteNorFlash` as mentioned in the previous comment. I don't find any reference in the PR explaining why it's not possible. I guess not all boards support multiple writes. Is there a solution to implement the trait only when it's supported? Should I open a separate issue or reopen this one?

Thanks!

huntc, 2022-01-08:
> Hi,
>
> AFAIU #337 did not implement `MultiwriteNorFlash` as mentioned in the previous comment. I don't find any reference in the PR explaining why it's not possible. I guess not all boards support [...]

I just implemented what I needed at the time. I’d suggest a new PR. Thanks.

ia0, 2022-01-08:
Sure thing! I created #373. I only implemented it for the nRF52 boards since I'm not familiar with the other ones.

commit 7a0be8a5 (2021-07-20): Merge #337

337: Provides common Nvmc functionality r=jonas-schievink a=huntc

The [embedded-storage](https://github.com/rust-embedded-community/embedded-storage) traits are implemented to provide a higher-level abstraction for reading and writing from/to flash storage. In the first instance, support is provided for the nRF52 boards and the nRF9160, but other boards should be relatively easy to support if required in the future.

Sample usage of writing to flash given some 32 bit aligned slice of bytes:

```rust
// Erase a page
let _ = nvmc.try_erase(0, 4096);

// Write the 32 bit aligned slic [... cut]
PR #337 (2021-07-20): Provides common Nvmc functionality
The [embedded-storage](https://github.com/rust-embedded-community/embedded-storage) traits are implemented to provide a higher-level abstraction for reading and writing from/to flash storage. In the first instance, support is provided for the nRF52 boards and the nRF9160, but other boards should be relatively easy to support if required in the future.

Sample usage of writing to flash given some 32 bit aligned slice of bytes:

```rust
// Erase a page
let _ = nvmc.try_erase(0, 4096);

// Write the 32 bit aligned slice
let _ = nvmc.try_write(0, some_u8_slice_on [... cut]