nrf-rs/nrf-hal #339: Can't use with onewire due to InputPin + OutputPin requirement

nevi-me, 2021-07-18:
When trying to use `nrf-hal-common` with OneWire (https://crates.io/crates/onewire), InputPin needs to be implemented for OutputPin.

I tried to implement this myself, but I seem to be a bit out of my depth :(

Example code is:

```rust
let peripherals = Peripherals::take().unwrap();

let pin1 = nrf52840_hal::gpio::p1::Parts::new(peripherals.P1);
let mut temp_pin = pin1.p1_08.degrade().into_open_drain_output(
    OpenDrainConfig::Disconnect0HighDrive1, Level::High
);
let mut wire = OneWire::new(&mut temp_pin, false);
```

This results in the error:

```rust
error[E0271]: type mismatch resolving `<nrf52840_hal::gpio::Pin<nrf52840_hal::gpio::Output<OpenDrain>> as nrf52840_hal::prelude::OutputPin>::Error == ()`
  --> src/main.rs:30:29
   |
30 | let mut wire = OneWire::new(&mut temp_pin, false);
   |                             ^^^^^^^^^^^^^ expected `()`, found enum `void::Void`
   |
   = note: required for the cast to the object type `dyn OpenDrainOutput<(), Error = (), Error = ()>`

error[E0277]: the trait bound `nrf52840_hal::gpio::Pin<nrf52840_hal::gpio::Output<OpenDrain>>: _embedded_hal_digital_InputPin` is not satisfied
  --> src/main.rs:30:29
   |
30 | let mut wire = OneWire::new(&mut temp_pin, false);
   |                             ^^^^^^^^^^^^^ the trait `_embedded_hal_digital_InputPin` is not implemented for `nrf52840_hal::gpio::Pin<nrf52840_hal::gpio::Output<OpenDrain>>`
   |
   = note: required because of the requirements on the impl of `nrf52840_hal::prelude::InputPin` for `nrf52840_hal::gpio::Pin<nrf52840_hal::gpio::Output<OpenDrain>>`
```

jonas-schievink, 2021-07-20:
I've opened https://github.com/nrf-rs/nrf-hal/pull/340 which should resolve this

unrelentingtech, 2021-08-09:
Previously: https://github.com/nrf-rs/nrf51-hal/pull/25

Finomnis, 2022-09-23:
I've opened #401 which should finally resolve this.