nrf-rs/nrf-hal #360: UART on nrf52840 blocks

1c3t3a, 2021-10-12:
Hello👋 I am currently trying to use the UART connection on the nrf52840-dongle. I try to do UART over the pins 0_15 and 0_20, but I am struggling as the UART write function hangs forever on waiting till the transmission is finished. The board is set up and also has an allocator configured at the point of sending via UART. This is the loop where my program hangs:
https://github.com/nrf-rs/nrf-hal/blob/f99d9371661aab76693aa1a8110328ebabbc89ec/nrf-hal-common/src/uarte.rs#L186-L188

My code for setting up the connection looks like this:

```rust
let p = pac::Peripherals::take().unwrap();

    let (uart0, cdc_pins) = {
        let p0 = gpio::p0::Parts::new(p.P0);
        (
            p.UARTE0,
            uarte::Pins {
                txd: p0.p0_15.into_push_pull_output(gpio::Level::High).degrade(),
                rxd: p0.p0_20.into_floating_input().degrade(),
                cts: Some(p0.p0_17.into_floating_input().degrade()),
                rts: Some(p0.p0_13.into_push_pull_output(gpio::Level::High).degrade()),
            },
        )
    };

    let uarte_conn = Uarte::new(
        uart0,
        cdc_pins,
        uarte::Parity::EXCLUDED,
        uarte::Baudrate::BAUD115200,
    )
```

I don't think that this is an issue with the library but rather with my usage of it, I just didn't really know where to ask else :)

1c3t3a, 2021-10-14:
Solved it on my own!

closed 2021-10-14 by 1c3t3a

Richardn2002, 2025-03-02:
How did you solve that? My code is also stuck at the same place when running the hello world example, and my current theory is I may need to properly set up the USB peripheral for the serial peripheral to properly complete the transactions (I am using an Arduino Nano 33 BLE Sense Rev 2 board). However I haven't figure out how to correctly setup USB either. If I ever get to fix this, I may update this comment.