nrf-rs/nrf-hal #367: nRF51822 Support

netthier, 2021-12-16:
Hey, is any support for the nRF51822, used by the Calliope Mini, planned?
If not, how hard would it be to implement?

jonas-schievink, 2021-12-16:
We don't currently have a PAC for that chip, so you'd have to add one to https://github.com/nrf-rs/nrf-pacs. Then, adding HAL support should be relatively straightforward (you can check https://github.com/nrf-rs/nrf-hal/pull/148 for an example PR adding support for a new chip).

netthier, 2021-12-16:
Do you know where I could find the SVD file for that chip? I tried downloading the MDK, and while it does contain SVD's for a lot of chips, there are only the following, relatively short, files for the 51822:

- `nrf51822_peripherals.h`
- `nrf51822_xxaa.ld`
- `nrf51822_xxab.ld`
- `nrf51822_xxac.ld`

Furthermore, the header file contains the line
`/* This file is deprecated */`
How would I generate a PAC crate for this chip?

jonas-schievink, 2021-12-17:
Oh, nRF5**1**822, that's already supported by the `nrf51-pac` and `nrf51-hal`. I thought it said nRF5**2**822, which does not seem to exist.

netthier, 2021-12-17:
Ah, that's awesome. Thanks!

closed 2021-12-17 by netthier