nrf-rs/nrf-hal #414: Order of operations make I²S tx() tricky to use

nospam2678, 2022-11-24:
Having spent quite some time on attempting to use the I²S API, I've come to question the _tx()_ method. It is inergonomic at best, or possibly even severely mis-designed.

When using the lower level methods _set_buffersize()_, _set_tx_ptr()_ and _start()_ are required to successfully initiate an I²S transfer. At least on the nRF52840.

Please consider figure 48 in "6.11.2 Transmitting and receiving" of [nRF52840_PS_v1.7](https://infocenter.nordicsemi.com/pdf/nRF52840_PS_v1.7.pdf). As is illustrated and described in the Product Specification, _RXD.PTR_ and _MAXCNT_ are to be set prior to triggering _TASKS_START_.

Yet it is necessary to call _start()_ prior to _tx()_, because the latter is designed to hide away the _I2S_ handle as a private variable inside _Transfer_ until completion. Undesired results are achieved for playing the first buffer when calling start with an invalid pointer and buffersize. That can be addressed by preparing by using _set_tx_ptr()_ for the very first transfer, but
that seems lika a bit of an unexpected requirement. To my eyes, it appears the documentation states that the value of _MAXCNT_ at the time of calling _TASKS_START_ is the one which will be used. With that being the case, it does not make much sense that _tx()_ [updates the register](https://github.com/nrf-rs/nrf-hal/blob/eb3b96b/nrf-hal-common/src/i2s.rs#L350) as it should had already have been done.

One might also take notice of that the [pull request](https://github.com/embassy-rs/embassy/pull/1049) for adding I²S to embassy [sets the pointer](https://github.com/chris-zen/embassy/blob/15a9324/embassy-nrf/src/i2s.rs#L662) prior to [starting the transfer](https://github.com/chris-zen/embassy/blob/15a9324/embassy-nrf/src/i2s.rs#L666). I.e. that implementation does the operations in the order documented in the Product Specification.

I'm not sure what the right fix here is. Either one could rip out all of the high level API methods, as they are only suitable for creating buggy code. Or one could make an attempt at fixing them. Making _tx()_ responsible for triggering _TASKS_START_ would be one possible change, but it is not immediately obvious to me what implications that would have. Start should likely only be triggered for the first buffer transmitted, so it would mean some kind of state needs to be maintained.

It is possible that _rx()_ and _transfer()_ have similar issues, but I have only looked at _tx()_.

nospam2678, 2022-11-26:
To clarify the above. I am successful in getting I²S started, but do not get correct results for the first buffer.

For context, my use case involves playing very short chirps. When the entire waveform is only a couple of hundred milliseconds, it's kind of crucial that every buffered byte actually gets played.

ninjasource, 2022-11-28:
I am not sure if I have used the I2S library as designed but I only ever called `start()` once on application startup and then did the actual `tx()` call inside the I2S interrupt handler itself. If there is no audio data then I `tx()` zeros for silence. I used a double buffer to queue up the next frame of audio to play without clobbering what is currently being DMA copied to the I2S device. Here is an example: https://github.com/ninjasource/nrf52840-dk-i2s-demo

This has the downside of introducing some latency that may not be acceptable for your use case though.