nrf-rs/nrf-hal #423: Support for embedded-hal v1.0

eflukx, 2023-07-21:
What are the thoughts on supporting/adopting the upcoming Embedded HAL v1.0 (currently in alpha stage)? 
Is the plan to wait until a proper v1.0 is release or would it make sense to have a support for the alpha versions in e.g. a separate branch of nrf-hal?

diondokter, 2023-07-21:
To be honest, this HAL isn't really maintained anymore. So unless a new maintainer steps up, this HAL may never support the new embedded-hal traits.

If you need the new traits, then I can recommend using `embassy-nrf`. It has pretty much all the functionality this HAL also has and is actively maintained. Don't worry about async, you don't need to use it.

eflukx, 2023-08-07:
Hmm.. yeah figured that given the last commit was done close to a year ago... To me this repo seemed the canonical HAL implementation for nRF-type hardware. I guess embassy(-nrf) is the way to go..

closed 2023-08-07 by eflukx

Yatekii, 2023-08-07:
I think we even discussed archiving the org somewhere :)