nrf-rs/nrf-hal #426: nrf51 uart write blocks in qemu

JoshMcguigan, 2023-09-04:
Hello! I am following along the [rust discovery uart example](https://docs.rust-embedded.org/discovery/microbit/07-uart/send-a-single-byte.html) by running the microbit v1 (nrf51) in [qemu](https://www.qemu.org/2019/05/22/microbit/). 

```
$ cargo build --features v1 --target thumbv6m-none-eabi

$ qemu-system-arm -machine microbit -display none -serial stdio -s -kernel target/thumbv6m-none-eabi/debug/uart
```
At first I thought the qemu emulation might be at fault, but I found [this forum post](https://devzone.nordicsemi.com/f/nordic-q-a/18914/how-does-nrf5-s-nrf_uart0--events_txdrdy-works) where they seem to see a similar problem on real hardware with code that matches the uart write impl here:

https://github.com/nrf-rs/nrf-hal/blob/939c0175dbbebdb9457a492a9912b755cc56737c/nrf-hal-common/src/uart.rs#L134-L141

This check for `txdrdy` hangs when trying to send the first byte. The reference manual says the `txdrdy` event is triggered only when a byte is actually sent. That aligns with the findings in the forum post that show very similar code blocking indefinitely.

Unfortunately I don't have real hardware to test this on at the moment.

kaidokert, 2025-05-05:
There's a workaround for this - after configuring the UART, simply write a dummy byte via PAC.

```
    let mut actual_uart = Uart::new(
        pac_uart,
        pins,
        uart::Parity::EXCLUDED,
        uart::Baudrate::BAUD115200,
    );
    // Write one byte to get txrdy to be set
    unsafe {
        let uart0 = &*hal::pac::UART0::ptr();
        uart0.txd.write(|w| w.txd().bits(b' '));
    }

```

The code in Uart::new actually already does that before configuring the peripheral, and that works on real hardware. But QEMU emulation is slightly different and it it ignores everything written to it before `.enabled()` bit gets set.

BartMassey, 2025-05-05:
The micro::bit v1 version of this book is pretty much out of maintenance at this point. That said, we would take a fix if it works on hardware without an unnecessary byte sent out on the UART. Bonus points if it also works on QEMU. Someone with a micro::bit v1 should test a proposed fix on both UART and QEMU.

BartMassey, 2025-05-05:
Yeah, thanks to @kaidokert for pointing out the issue. I think this should probably be fixed in QEMU, not here, but contrary opinions are welcome. Don't want to put a dummy byte in the UART after enabling it, as it will presumably go out on the wire.

JoshMcguigan, 2025-05-10:
Thanks @kaidokert, I agree this sounds like a QEMU issue. I'm going to close this issue.

closed 2025-05-10 by JoshMcguigan