nrf-rs/nrf-hal #483: Radio on nRF52832

arnobert, 2024-07-26:
Hello,

I would like to use the radio peripheral on the nRF52 DK board, just like on the nRF52840 dongle, but the lib is not available, yet. Is there any special reason for that?

Thank you!

BartMassey, 2024-07-26:
Can Embassy's `nrf-softdevice` work for you here?

arnobert, 2024-07-27:
I am using RTIC and would prefer to use the HAL, just like on the 52840.

BartMassey, 2024-07-27:
Sorry, I'm new to this. Perhaps @Dirbaio would be able to help? You might also want to try on [#rust-embedded:matrix.org](https://matrix.to/#/!BHcierreUuwCMxVqOf:matrix.org) for someone more knowledgeable than I.

In the meantime, can you point me to example code for using nRF Softdevice with `embedded-hal` on the 52840? I'm curious.

Dirbaio, 2024-07-27:
There's examples [here](https://github.com/embassy-rs/nrf-softdevice/tree/master/examples), both for nrf52840 and nrf52832.

> I am using RTIC and would prefer to use the HAL, just like on the 52840.

both `nrf52840-hal` and `embassy-nrf` *are* HALs, you can use one or the other with RTIC. There's an [example](https://github.com/embassy-rs/embassy/tree/main/examples/nrf52840-rtic) for it.

BartMassey, 2024-07-27:
@Dirbaio thanks!

@arnobert is there code somewhere for using nRF SoftDevice with `nrf52840-hal`? I couldn't find any.

arnobert, 2024-07-28:
@BartMassey I don't know. I simply used: https://github.com/nrf-rs/nrf-hal/blob/master/nrf-hal-common/src/ieee802154.rs

arnobert, 2025-02-10:
> [@Dirbaio](https://github.com/Dirbaio) thanks!
> 
> [@arnobert](https://github.com/arnobert) is there code somewhere for using nRF SoftDevice with `nrf52840-hal`? I couldn't find any.

Hi,
a late reply, but yes, I have finally managed to put a demo on git:

https://gitlab.com/DL2SSB/nrf_radio/

Would be great, if that worked with the nrf52832 as well!
Cheers!